rwaldron/johnny-five #1739: Adafruit VCNL4010 Proximity/Light sensor & Johnny-five

diegorivasn, 2021-08-20:
Hi guys, checking the proximity sensors currently supported by Johnny-Five, I can't see a controller that could work with the [Adafruit VCNL4010 Proximity/Light sensor](https://www.adafruit.com/product/466). Is there a way to control this component using Johnny-five + Firmata + Arduino without having to modify the libraries?  Thanks in advance!

dtex, 2021-08-20:
It doesn't support it yet, but contributions are always welcome. It's a little tricky since it comprises multiple sensors, but there is precedent. I think I'd look at the MPU6050 for a good example of how to organize it. This would be the first SiP with proximity and light components.

diegorivasn, 2021-08-23:
Thanks for your quick comment, I'll take a look at it