rwaldron/johnny-five #1775: Dual Servo motors conflict

J040M, 2022-01-15:
Hey everyone.

I'm having a bit of an issue using 2 servo motors.
I've tried to declare them in all the ways described in the examples but only 1 of them ever works (always the one initialized first).

I've also had the problem that when injecting the servos into repl 1 of the servos would reply to commands sent to both servos while the other would do nothing.

Has anyone else had servo conflicts like this?

Cheers

dtex, 2022-01-15:
Hi @J040M 

Yes, I've seen that but only when there are other devices involved. Can I see your code, and ask what microcontroller are you using?

J040M, 2022-01-19:
Hey @dtex 

Yes, I'm using **arduino uno**. I've tried quite a few versions of the code. 

ex 1
`
const { Board, Servo, Servos } = require("johnny-five");
const board = new Board();

board.on("ready", async () => {
  const servos = new Servos([12, 13]);

  servos.center();

  board.repl.inject({ servos });
});`

ex 2
`
board.on("ready", async () => {
  const servo1 = new Servo({
    pin: 12,
    range: [0, 180],
    fps: 100,
  });

  const servo2 = new Servo({
    pin: 13,
    range: [0, 180],
    fps: 100,
  });

  const servos = new Servos([servo1, servo2]);

  servos.center();
});
`

Cheers