rwaldron/johnny-five #1777: Error: Cannot find module 'firmata'

dasamepu, 2022-02-06:
Hello, I was trying to run the example but when I write in the command-line "node app.js" it shows me "Error: Cannot find module 'firmata'".

I'm not sure but I think it's because the file "C:\Users\user\AppData\Local\Temp\build123.tmp\StandardFirmata.cpp.hex" doesn't exist in my PC and in the Aduino IDE console log at bottom it shows me this file: "C:\Users\user\AppData\Local\Temp\arduino_build_560749/StandardFirmata.ino.elf".

Maybe that is the reason beacause is not working or not, but I'll be really thankfull if you help me.
[image]

dtex, 2022-02-06:
That particular error message isn't related to the Arduino code, it's just saying it can't find the node.js module "firmata". That should have been installed when you installed johnny-five so it's a bit of a mystery. Anything non-standard about your install process? Were there any errors when you ran `npm i johnny-five`? I assume this is running on Windows and connected to an Arduino via USB correct? If so try `npm i firmata` from within the `proteusnode` directory.

dasamepu, 2022-02-07:
Yes, when I install johnny-five it looks like it wasn't intalled properly.
[image]

And when I install firmata it also shows me errors and says that I have should have to install python, but why if I won't use python?
[image]
[image]

Amaranthusss, 2022-04-19:
I have [the same problem](https://github.com/rwaldron/johnny-five/issues/1786), at this moment could only put manually `firmata` library data to `node_modules` and it helps.

WyattSanders, 2023-11-14:
> I have [the same problem](https://github.com/rwaldron/johnny-five/issues/1786), at this moment could only put manually `firmata` library data to `node_modules` and it helps.

I've found the solution here:
https://github.com/rwaldron/johnny-five/issues/1807#issuecomment-1809561715