rwaldron/johnny-five #1779: How to stop stepper motor programmatically?

berzins92, 2022-03-01:
Hi there!
I have problems stoping stepper driver programmatically in case it triggers end-stop switch.

I can't find API reference to this. Anyone has any idea how to do this?

I am using: `five.Stepper.TYPE.DRIVER`

dtex, 2022-03-24:
@berzins92 Johnny-Five supports the first version of stepper built into firmata. That version does not allow for stopping a stepper programmatically. 

There is a newer version of stepper called AccelStepperFirmata that does support stopping programmatically. You can get a custom build of firmata over at http://firmatabuilder.com/ that includes the newer version, and you can [see how it's used by firmata.js in this example](https://github.com/firmata/firmata.js/blob/master/packages/firmata.js/examples/stepper-accel.js). 

The [firmata.js documentation for AccelStepper](https://github.com/firmata/firmata.js/tree/master/packages/firmata.js#accelstepperfirmata) is pretty clear, and the API is much more "Johnny-Five" like than other firmata features. You can talk directly to firmata in a Johnny-Five program by accessing the 'board.io' property (assuming you called it `board`).

berzins92, 2022-04-03:
Thanks @dtex.
`baord.io.accelStepperStop(1)`

where 1 is stepper id.

closed 2022-04-03 by berzins92