rwaldron/johnny-five #1780: Old serialport (< 10.x.x) usage

lucas-sg, 2022-03-01:
I believe that an old version of serialport is being used. The current serialport (10.3.0) doesn't make use of @serialport/bindings, instead it uses @serialport/bindings-cpp. I'd love to use the current serialport version as it's compatible with the latest Electron version and I'm not forced to use Electron v12 or below.

Is there any way I could help updating this package so that this can be solved?

code-valor, 2022-03-22:
Same. I get error that saying "serialport.list is not a function". I'm getting cold from this package...

MiguelRavagnani, 2022-03-29:
Still an issue.

I was able to make It work by downgrading serialport to 9.2.8

iamisti, 2022-11-28:
wow this is still open?

closed 2023-01-23 by dtex