rwaldron/johnny-five #1786: Cannot find module 'firmata' - works only at manual copy to node_modules

Amaranthusss, 2022-04-19:
Hello there,

It looks like there is a some bug at johnny-five library's dependencies (I am not sure, please verify).

A long time ago I prepared simple NodeJS script to test NodeMcu V3 board and controlling it with the firmata firmware. And that worked but this time I could not install correctly Johnny-Five. So I made a new project with only 3 dependencies for a test:
`
    "etherport-client": "^0.1.4",
    "johnny-five": "1.5.0",
    "keypress": "^0.2.1",
`
And when I start the simple script shown below I got error.
Script: 
`
const { EtherPortClient } = require("etherport-client")
const five = require("johnny-five")
const keypress = require("keypress")

const board = new five.Board({
	port: new EtherPortClient({
		host: "192.168.8.122",
		port: 3030,
	}),
	repl: false,
})
`
Error:
`
node:internal/modules/cjs/loader:936
  throw err;
  ^

Error: Cannot find module 'firmata'
Require stack:
- C:\Users\Amaranthuss\Desktop\firmata-fix\node_modules\johnny-five\lib\board.js
- C:\Users\Amaranthuss\Desktop\firmata-fix\node_modules\johnny-five\lib\accelerometer.js
- C:\Users\Amaranthuss\Desktop\firmata-fix\node_modules\johnny-five\lib\johnny-five.js
- C:\Users\Amaranthuss\Desktop\firmata-fix\index.js
`

I tried to install firmata but it is not possible.
`
C:\Users\Amaranthuss\Desktop\firmata-fix>npm i firmata
npm ERR! code 1
npm ERR! path C:\Users\Amaranthuss\Desktop\firmata-fix\node_modules\@serialport\bindings
npm ERR! command failed
npm ERR! command C:\WINDOWS\system32\cmd.exe /d /s /c prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@8.4.1
npm ERR! gyp info using node@16.14.2 | win32 | x64
npm ERR! gyp info find Python using Python version 3.10.4 found at "C:\Python310\python.exe"
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS msvs_version not set from command line or npm config
npm ERR! gyp ERR! find VS VCINSTALLDIR not set, not running in VS Command Prompt
npm ERR! gyp ERR! find VS checking VS2019 (16.11.32413.511) found at:
npm ERR! gyp ERR! find VS "C:\Program Files (x86)\Microsoft Visual Studio\2019\BuildTools"
npm ERR! gyp ERR! find VS - found "Visual Studio C++ core features"
npm ERR! gyp ERR! find VS - found VC++ toolset: v142
npm ERR! gyp ERR! find VS - missing any Windows SDK
npm ERR! gyp ERR! find VS could not find a version of Visual Studio 2017 or newer to use
npm ERR! gyp ERR! find VS looking for Visual Studio 2015
npm ERR! gyp ERR! find VS - not found
npm ERR! gyp ERR! find VS not looking for VS2013 as it is only supported up to Node.js 8
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! find VS **************************************************************
npm ERR! gyp ERR! find VS You need to install the latest version of Visual Studio
npm ERR! gyp ERR! find VS including the "Desktop development with C++" workload.
npm ERR! gyp ERR! find VS For more information consult the documentation at:
npm ERR! gyp ERR! find VS https://github.com/nodejs/node-gyp#on-windows
npm ERR! gyp ERR! find VS **************************************************************
npm ERR! gyp ERR! find VS
npm ERR! gyp ERR! configure error
npm ERR! gyp ERR! stack Error: Could not find any Visual Studio installation to use
npm ERR! gyp ERR! stack     at VisualStudioFinder.fail (C:\Users\Amaranthuss\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:122:47)
npm ERR! gyp ERR! stack     at C:\Users\Amaranthuss\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:75:16
npm ERR! gyp ERR! stack     at VisualStudioFinder.findVisualStudio2013 (C:\Users\Amaranthuss\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:363:14)
npm ERR! gyp ERR! stack     at C:\Users\Amaranthuss\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:71:14
npm ERR! gyp ERR! stack     at C:\Users\Amaranthuss\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\find-visualstudio.js:384:16
npm ERR! gyp ERR! stack     at C:\Users\Amaranthuss\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\util.js:54:7
npm ERR! gyp ERR! stack     at C:\Users\Amaranthuss\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\lib\util.js:33:16
npm ERR! gyp ERR! stack     at ChildProcess.exithandler (node:child_process:406:5)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:526:28)
npm ERR! gyp ERR! stack     at maybeClose (node:internal/child_process:1092:16)
npm ERR! gyp ERR! System Windows_NT 10.0.19044
npm ERR! gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Amaranthuss\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd C:\Users\Amaranthuss\Desktop\firmata-fix\node_modules\@serialport\bindings
npm ERR! gyp ERR! node -v v16.14.2
npm ERR! gyp ERR! node-gyp -v v8.4.1
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\Amaranthuss\AppData\Local\npm-cache\_logs\2022-04-19T21_38_42_386Z-debug-0.log
`
I saw a like [topic](https://github.com/rwaldron/johnny-five/issues/1267) but in my case manual installation did not help.

**How I did solve the problem:**
Then I downloaded zip files [firmata](https://github.com/firmata/firmata.js) library (from packages folder) and I copied them to my project `node_modules` (I had to rename `firmata.js` folder to `firmata`). 
Additionally I had to install serialport because there was a next one error about missing module. But in this case it could be installed by `npm i serialport`.

Next, I added manually to dependencies list 
`
"firmata": "^2.3.0",
"firmata-io": "^2.3.0"
`
but still there is not possible to install them. I got the same error message as above (using node-gyp).

I tested at second PC to isolate my computer software issues. There was a same problem. 

Did I do something wrong or that is kind of bug? What can I do to skip copy `firmata` at each `package.json` changes? :(

Thank you in advance

System: Windows 10

kirill321592, 2022-04-20:
Hello try to install jhonny5 wih yarn and then install seriallport 9.2.8

Amaranthusss, 2022-04-20:
Lol that worked but only at serialport v9.2.8 as you said. 
```
    "etherport-client": "^0.1.4",
    "johnny-five": "^2.1.0",
    "keypress": "^0.2.1",
    "serialport": "9.2.8"
```
But again there is the same error as above at installation - `gyp ERR!`. 
I did a test two tests:
1. I installed `johnny-five` as the last one and I got error only once - at `yarn add johnny-five`.
2. And next I changed order of libraries installation: `johnny-five` as first and then the rest of. I got `gyp ERR!` at each next added library after `johnny-five`. Anyway that has been installed and works. 

At 10.4.0 (the newest one) did not.
```
It looks like serialport didn't install properly.
More information can be found here https://serialport.io/docs/guide-installation
The result of requiring the package is: undefined
Error: Cannot find module '@serialport/bindings'
Require stack:
- C:\Users\Amaranthuss\Desktop\yarn\node_modules\firmata\node_modules\serialport\lib\index.js
- C:\Users\Amaranthuss\Desktop\yarn\node_modules\firmata\lib\com.js
- C:\Users\Amaranthuss\Desktop\yarn\node_modules\firmata\lib\firmata.js
- C:\Users\Amaranthuss\Desktop\yarn\node_modules\johnny-five\lib\board.js
- C:\Users\Amaranthuss\Desktop\yarn\node_modules\johnny-five\lib\accelerometer.js
- C:\Users\Amaranthuss\Desktop\yarn\node_modules\johnny-five\lib\johnny-five.js
- C:\Users\Amaranthuss\Desktop\yarn\index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (C:\Users\Amaranthuss\Desktop\yarn\node_modules\firmata\node_modules\serialport\lib\index.js:2:17)
    at Module._compile (node:internal/modules/cjs/loader:1103:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1157:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    'C:\\Users\\Amaranthuss\\Desktop\\yarn\\node_modules\\firmata\\node_modules\\serialport\\lib\\index.js',
    'C:\\Users\\Amaranthuss\\Desktop\\yarn\\node_modules\\firmata\\lib\\com.js',
    'C:\\Users\\Amaranthuss\\Desktop\\yarn\\node_modules\\firmata\\lib\\firmata.js',
    'C:\\Users\\Amaranthuss\\Desktop\\yarn\\node_modules\\johnny-five\\lib\\board.js',
    'C:\\Users\\Amaranthuss\\Desktop\\yarn\\node_modules\\johnny-five\\lib\\accelerometer.js',
    'C:\\Users\\Amaranthuss\\Desktop\\yarn\\node_modules\\johnny-five\\lib\\johnny-five.js',
    'C:\\Users\\Amaranthuss\\Desktop\\yarn\\index.js'
  ]
}

C:\Users\Amaranthuss\Desktop\yarn\node_modules\firmata\lib\com.js:61
    throw "Missing serialport dependency";
    ^
Missing serialport dependency
(Use `node --trace-uncaught ...` to show where the exception was thrown)
```

That's weird.

jimmyfio, 2022-04-23:
I had a similar problem with firmata and serialport when I tried to install my project on a new machine from a github clone.  I couldn't figure out the problem, but at some point I realize that the new machine was running the most current version of Node, while the machine I built my project on was running an older version. I downgraded my version of Node to match the old version I built it with and everything installed perfectly.

GitForArtTech, 2022-10-26:
Hi there,
This is work for me:
1. downgrade node version to v14.17.6 (npm v6...)
2. install Visual Studio Build Tools 2019 (cause i faced on some gyp error)
3. delete original node_modules & package-lock.json
4. package.json only depence on "johnny-five":"^2.1.0"
5. npm install
and everything seems to work successfully.