rwaldron/johnny-five #1788: Old code is running instead of new code

VeqsaGit, 2022-06-08:
I tried the LED blinking example,
then tried the rainbow example... but instead of it working, the LED blinking code runned.
i tried debugging by adding console.log's in both files, but the old code does not run even tho edited,
and the new code runs but not inside of the board.on("ready")!

I even tried resetting, but when i touch the reset button on the arduino (uno) (doesnt matter for how long) and release, it reruns the code:
so the LED starts blinking again...

i also tried uploading code and all other sort of things like restarting my pc,
but it is stuck at the first example... how should i fix this?

manno-xx, 2022-06-16:
Resetting the Arduino only restarts what's on there.
When you say 'uploading code' I assume using the arduino IDE and other code than the standard Blink example?
What if you power the arduino by means of a power bank or other computer?
It cannot possibly be that the stnadard Blink example is running on the Arduino

VeqsaGit, 2022-06-25:
No then it does nothing...
But when reconnecting to the pc, it still is not reset.

I want to change my code, so please lmk