rwaldron/johnny-five #1789: TypeError: serialport.list is not a function (board.js line 42)

manno-xx, 2022-06-16:
I'm trying to connect my macbook (year 2018, running monterey) to an arduino. When running the standard blink example as a test, I get the error:

`
/Users/manno/node_modules/johnny-five/lib/board.js:42
    serialport.list().then(results => {
               ^

TypeError: serialport.list is not a function
    at Board.detect (/Users/manno/node_modules/johnny-five/lib/board.js:42:16)
    at new Board (/Users/manno/node_modules/johnny-five/lib/board.js:291:23)
    at Object.<anonymous> (/Users/manno/[some more path]/NodeWebSocketServer/withFirmata.js:2:13)
`

Which points to the one-before-last line in:
`
let serialport = serialport;

    /* istanbul ignore next */
    if (parseFloat(process.versions.nw) >= 0.13) {
      serialport = require("browser-serialport");
    } else {
      serialport = require("serialport");
    }

    // console.log(require);
    // Request a list of available ports, from
    // the result set, filter for valid paths
    // via known path pattern match.
    serialport.list().then(results => {
      const portPaths = results.reduce((accum, result) => {

`

npm list gives me:

├── johnny-five@2.1.0
├── serialport@10.4.0
└── ws@8.7.0

Anyone an idea how to solve this?

Kind regards,
Manno

manno-xx, 2022-06-16:
Yes, sorry :)

It is just the standard blink from the homepage:

```
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function(){
    var led = new five.Led(13);
    led.blink(500);
});
```

manno-xx, 2022-07-05:
Btw: I removed some stuff to make sure there were no wrong libraries referenced or something. Then I ran the following (with the same result):

```
manno@Manno NodeWebSocketServer % node withFirmata.js 
node:internal/modules/cjs/loader:936
  throw err;
  ^

Error: Cannot find module '@serialport/bindings'
Require stack:
- /Users/manno/node_modules/johnny-five/node_modules/serialport/lib/index.js
- /Users/manno/node_modules/johnny-five/lib/board.js
- /Users/manno/node_modules/johnny-five/lib/accelerometer.js
- /Users/manno/node_modules/johnny-five/lib/johnny-five.js
- /Users/manno/Library/CloudStorage/OneDrive-HanzehogeschoolGroningen/P0.0 Development/Web Dev/Server/NodeWebSocketServer/withFirmata.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:933:15)
    at Function.Module._load (node:internal/modules/cjs/loader:778:27)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/Users/manno/node_modules/johnny-five/node_modules/serialport/lib/index.js:2:17)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/Users/manno/node_modules/johnny-five/node_modules/serialport/lib/index.js',
    '/Users/manno/node_modules/johnny-five/lib/board.js',
    '/Users/manno/node_modules/johnny-five/lib/accelerometer.js',
    '/Users/manno/node_modules/johnny-five/lib/johnny-five.js',
    '/Users/manno/Library/CloudStorage/OneDrive-HanzehogeschoolGroningen/P0.0 Development/Web Dev/Server/NodeWebSocketServer/withFirmata.js'
  ]
}
manno@Manno NodeWebSocketServer % npm install serialport

added 2 packages, removed 10 packages, changed 4 packages, and audited 37 packages in 20s

12 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
manno@Manno NodeWebSocketServer % node withFirmata.js   
/Users/manno/node_modules/johnny-five/lib/board.js:42
    serialport.list().then(results => {
               ^

TypeError: serialport.list is not a function
    at Board.detect (/Users/manno/node_modules/johnny-five/lib/board.js:42:16)
    at new Board (/Users/manno/node_modules/johnny-five/lib/board.js:291:23)
    at Object.<anonymous> (/Users/manno/Library/CloudStorage/OneDrive-HanzehogeschoolGroningen/P0.0 Development/Web Dev/Server/NodeWebSocketServer/withFirmata.js:2:13)
    at Module._compile (node:internal/modules/cjs/loader:1105:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1159:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:77:12)
    at node:internal/main/run_main_module:17:47
```

Cerceis, 2022-08-10:
I have the same problem. Running on Macbook pro 2019. With the example code provided.

sminliwu, 2022-10-14:
I was having this same problem when running the blink example code. I'm using VSCode on Windows 10. After a couple hours of banging my head on the wall, I got the example code to run. The things I did:

1.  I looked at the `serialport` definition in `node_modules/serialport` and it turned out that the exported name is `SerialPort`. I modified `board.js` (`node_modules\johnny-five\lib\board.js`) starting at line 29, until the `serialport.list()` call:

```
    let serialport;

    /* istanbul ignore next */
    if (parseFloat(process.versions.nw) >= 0.13) {
      let { SerialPort } = require("browser-serialport");
    } else {
      let { SerialPort } = require("serialport");
      serialport = SerialPort;
    }

    // console.log(require);
    // Request a list of available ports, from
    // the result set, filter for valid paths
    // via known path pattern match.
    serialport.list().then(results => {
```

2. TL;DR 
    a. Install the `firmata-io` package by running `npm install firmata-io`. 
    b. Change the first line of the `connect(portOrPath, callBack)` function that requires the Firmata package: 
 ``` 
    connect(portOrPath, callback) {
        const IO = require("firmata-io").Board;
 ```
After the serial port error was gone, I got this new error that the package `firmata` could not be found. I tried to install with `npm install firmata` but kept getting an error with node-gyp or Python build error or whatever, and that was getting way over my head with what I know about Node.JS. But it seems like there's an alternate version of the Firmata package. Still trying to get the example code to work, though. It seems to not finish executing.

Pavelkovo, 2022-11-07:
Faced the same problem. An example is a flashing LED. Clean installation.
I also noticed that the windows-build-tools were moved to the archive...

renickbell, 2022-11-08:
I'm having the same problem on two Windows 11 machines and one Mac. With Linux, it seems fine.

dwainH, 2023-05-07:
I am also having this issue on Windows 11

Markusme81, 2023-05-10:
I have the same issue and I don't know how to update the serialport.

zcrich, 2023-07-13:
I was getting the same `TypeError: serialport.list is not a function` error on MacOS 13.3.1 w/M1

I re-installed node and npm using brew and the issue resolved.
```
$ brew install node
$ brew install npm
```

From a project folder:
```
$ npm init
$ npm install johnny-five
```

Johnny-Five's `LED-Blink` example (saved to example/blink.js) working fine now on Arduino Uno running `StandardFirmataPlus` sketch :-) 

```
$ node example/blink.js
1689205744025 Available /dev/tty.usbmodem1301
1689205744031 Connected /dev/tty.usbmodem1301
1689205745713 Repl Initialized
>> Ready!

(To exit, press Ctrl+C again or Ctrl+D or type .exit)
>>

```

ekimbasoglu, 2023-09-19:
@zcrich did the same exact steps, still having the same issue by johnny-five on board.js:42:16 and 291:23, have the same device M1A

`node_modules/johnny-five/lib/board.js:42
    serialport.list().then(results => {
               ^

TypeError: serialport.list is not a function`

Alexei-MD, 2023-10-04:
Hey you still have the probleme, you have a solution for this issues ?

ekimbasoglu, 2023-10-04:
@Alexei-MD instead of using it I used serialPort natively, if you only use serialPort you won't encounter any issues, you can check out on my pinned repo how I used it

MkFoster, 2024-05-03:
I ran into the "serialport.list is not a function" error after installing Johnny-Five on a Windows machine.  In my case, the "serialport" and "firmata" libraries failed to build and install when I installed Johnny-Five. I think I didn't notice because the Johnny-Five package.json has those nested under "optionalDepencies." So, although they failed to build/install, it appeared the installation was fine.  I searched my node_modules folder and finally noticed they simply weren't there. So yeah, serialport.list() was not a function because serialport wasn't there!

The following steps fixed the problem in my case:
- Upgrade Python to the latest
- "pip install setuptools" at the command line
- "npm remove Johhny-Five" followed by "npm i Johhny-Five."

The Python "setuptools" package is needed to build some NodeJS libraries with native C++ code like [serialport](https://github.com/serialport/bindings-cpp/tree/main/src).  After completing these steps my NodeJS code could talk to my Arduino and I confirmed that "serialport" and "firmata" were installed under node_modules.

ekimbasoglu, 2024-05-03:
@MkFoster at this point I used this like below;
```
const SerialPort = require("serialport").SerialPort;
const Readline = require("@serialport/parser-readline");
```
I needed it for a tiny project and it is covered [here](https://github.com/ekimbasoglu/Smart-Weather-Link/blob/main/api/server.js)

Thank you for sharing how it might be caused and for the fix