rwaldron/johnny-five #1807: johnny-five not working

Sarthak-Arya, 2022-12-29:
```
const { Board, Led } = require("johnny-five");
const board = new Board({port:"COM3"});

board.on("ready", () => {

  // Create a standard `led` component instance
  const led = new Led(13);
  this.repl.inject({
    led: led
  });
  // "blink" the led in 500ms
  // on-off phase periods
  led.blink(1000);
});
```
Above is my code

[image]

Here is the output is not showing anything 
I am using Windows 11 x64 bit system

Jahson17, 2023-01-11:
Is it throwing an error like "cannot find module 'firmata' " ???

Or what's the result actually..

abdooo9, 2023-01-12:
> Is it throwing an error like "cannot find module 'firmata' " ???

I don't think so, but I'm facing this error, is there a solution?, I tried install firmata package but always fails: `npm ERR`

princeborn1999, 2023-09-07:
I'm facing same problem.

WyattSanders, 2023-11-11:
> Is it throwing an error like "cannot find module 'firmata' " ???
> 
> Or what's the result actually..

Error: Cannot find module 'firmata'

It's almost a year later and I'm having the same issue with no ability to find a solution.

tbaltrushaitis, 2023-11-11:
`npm i firmata`

tbaltrushaitis, 2023-11-11:
> ```
> const { Board, Led } = require("johnny-five");
> const board = new Board({port:"COM3"});
>
> board.on("ready", () => {
>
>   // Create a standard `led` component instance
>   const led = new Led(13); [...]

Your code shouldn't output anything. It should blink the default led on your board (of course if it is really on port 'COM3')

tbaltrushaitis, 2023-11-11:
Also I'm not sure if `this` would represent your board because of `() => {}`

WyattSanders, 2023-11-14:
> `npm i firmata`

This wouldn't fix it, and there would be a laundry list of problems that would follow. 
**I have however figured out the root cause for this issue and others**
People trying to run Johnny-five with a fresh install of Windows 10 or 11 and windows build tools 2022 CANNOT install johnny-five with "npm install" I have found that you need node version 13.14.0 or 14.17.0 (if you tried installing johnny-five with a newer version you need to delete your node_modules folder then change versions of node with nvm-windows) then install VS Build Tools 2019 (VS Build Tools will not work!) with C++ development. 

After this johnny-five can install successfully with npm install. (Antivirus may also need turned off, run from powershell with administrator privileges too)
[image]

reymesson1, 2023-12-07:
Remember also to turn off any Bluetooth connection, make sure that the package.json has into script: "firmata": "firmata-party uno --debug" and npm run firmata. And then node index.js

tbaltrushaitis, 2024-08-10:
> i tried so many ways to rectify this issue,but i can't
> 
> PS D:\J5> node index.js node:internal/modules/cjs/loader:1080 throw err; ^
> 
> Error: Cannot find module 'serialport' 

Install it as described [here](https://serialport.io/)