rwaldron/johnny-five #1817: johny-five initailization pb with ConfigurableFirmata

echavet, 2023-02-28:
I can manage to init johnny-five with the blink (helloWorld.js) example if I use the StandardFirmataPlus firmware on an arduino UNO but I can't manage to get it work with ConfigurableFirmata which is supposed to be supported in the doc.

I get an exception : 

>serialport/bindings/unixRead read error [Error: EAGAIN: resource temporarily unavailable, read] {
  errno: -35,
  code: 'EAGAIN',
  syscall: 'read'
} +0ms
  serialport/bindings/unixRead waiting for readable because of code: EAGAIN +0ms
  serialport/bindings/poller Polling for "readable" +1ms
  serialport/stream _write 1 bytes of data +2s
  serialport/binding-abstract write 1 bytes +2s
  serialport/bindings/unixWrite Starting write 1 bytes offset 0 bytesToWrite 1 +0ms
  serialport/bindings/unixWrite write returned: wrote 1 bytes +0ms
  serialport/bindings/unixWrite Finished writing 1 bytes +1ms
  serialport/stream binding.write write finished +2ms
  serialport/stream _write 3 bytes of data +0ms
  serialport/binding-abstract write 3 bytes +2ms
  serialport/bindings/unixWrite Starting write 3 bytes offset 0 bytesToWrite 3 +0ms
  serialport/bindings/unixWrite write returned: wrote 3 bytes +0ms
  serialport/bindings/unixWrite Finished writing 3 bytes +0ms
  serialport/stream binding.write write finished +1ms
1677596731635 Device or Firmware Error A timeout occurred while connecting to the Board. 

>Please check that you've properly flashed the board with the correct firmware.
See: https://github.com/rwaldron/johnny-five/wiki/Getting-Started#trouble-shooting

>If connecting to a Leonardo or Leonardo clone, press the 'Reset' button on the board, wait approximately 11 seconds for complete reset, then run your program again.  
Erreur: {"type":"error","timestamp":1677596731635,"class":"Device or Firmware Error","message":"A timeout occurred while connecting to the Board. 

Of course I've read the trouble-shooting section which indicate to use a StandardFirmata firmware. But if I do so, I cannot use the OneWire Thermometer feature of Johnny-five (which indicates me to use ConfigurableFirmata when I'm using StandardFirmata).

There is no log to help me to trouble shoot this issue!! 

[image]

echavet, 2023-03-01:
Precision: ConfigurableFirmata V 2.10.1 works well. So the pb is only with 3.0.0 version of ConfigurableFirmata

pgrawehr, 2023-04-02:
Interesting, I'm not aware of a protocol change that would affect the initialization sequence. And other client libraries seem to be unaffected.

echavet, 2023-04-02:
I'm trying to find out a solution to make them speak together.
However it's difficult to debug I found that with the help of chatGPT!
Envoyé par johnny-five : 249
Reçu par ConfigurableFirmata : 134

Débogage avec ConfigurableFirmata.

Model: GPT-4

Famille Chavet
en regardant les poids forts et faible et en imaginant les erreurs de swap
possibles est-ce que tu vois un lien entre 249 et 134 ?

ChatGPT:

Oui, il y a un lien entre 249 et 134 si l'on considère les représentations
binaires de ces deux nombres. Les voici en binaire sur 8 bits :

249 : 11111001
134 : 10000110
On peut observer que les 4 bits de poids fort de 249 (1111) sont inversés
pour donner les 4 bits de poids faible de 134 (0110), et les 4 bits de
poids faible de 249 (1001) sont inversés pour donner les 4 bits de poids
fort de 134 (1000).

Si l'on considère que les bits sont inversés (swap), on obtient :

249 : 11111001 -> 10000110
134 : 10000110 -> 11111001
Il semble donc que les bits soient inversés lors de la transmission des
données. Cela pourrait être dû à une mauvaise configuration de la
communication série (parité, bits de données, bits d'arrêt) ou à un
problème matériel.

Vérifiez les configurations de communication série côté Arduino et Node.js
pour vous assurer qu'elles correspondent. Normalement, les paramètres par
défaut pour la communication série sont :

Parité : aucune
Bits de données : 8
Bits d'arrêt : 1
Si les configurations correspondent déjà, essayez de changer de câble USB
ou de port USB sur votre ordinateur, car cela pourrait être dû à un
problème matériel.

pgrawehr, 2023-04-02:
Hmmm... I don't quite understand how that could help. The bit order certainly hasn't changed.

echavet, 2023-04-02:
I don't know how the reason yet. But I know for sure that one command is
sent and that it is not understood because bits are upside down !!

closed 2023-04-04 by echavet

pgrawehr, 2023-04-04:
@echavet Oh, yea. That's what did change. The default baudrate has changed from 57600 to 115200.

awong1900, 2023-04-06:
@echavet I think it is init problem on ConfigurableFirmata.ino. Can you follow this pr, and test it?