rwaldron/johnny-five #1819: Supporting NRF24L01

Jenesius, 2023-03-24:
I looked at the API provided by *Johnny-five*, but couldn't find any information on working with **radios**. Are there any examples of working with sensors like **NRF24L01**. In case I missed a mention of this in the documentation, please open my eyes. Thank you!

Jenesius, 2023-03-24:
```js
const five = require("johnny-five");
const NRF24 = require("nrf24");

const board = new five.Board();

board.on("ready", function() {
  const radio = new NRF24(this, {
    ce: 9,
    irq: 8,
    spi: {
      clock: 1000000,
      device: "/dev/spidev0.0"
    }
  });

  radio.begin();

  radio.printDetails(); 
});
```

dtex, 2023-03-24:
@Jenesius Johnny-Five doesn't have support for the nRF24, and I doubt it ever will. Even the [Nordic site](https://www.nordicsemi.com/products/nrf24-series) recommends using Bluetooth for new projects.

I always hate when someone responds to a request like this with "You should use X instead", but in this case maybe there is a better option. Can I ask what you are trying to do?