rwaldron/johnny-five #1820: Novice ask next how to upload js to hardware

zhou1591, 2023-05-16:
My highly skilled brothers ，I would like to ask how I can upload js to the hardware, that is, I connect a battery to the hardware, and then it can execute the logic of js I upload。

lbineau, 2023-08-16:
Hello @zhou1591 ,
As far as I know you always need a computer to be able execute nodeJS in order to be able to run johnny-five. 
It is not code that you can compile and put on Arduino with a battery.
If you want something compact you'll need something like a raspberry pi zero in addition to your Arduino to execute johnny-five.

scottgonzalez, 2023-10-05:
This requires an on-board linux environment. See the [Platform Support page](https://johnny-five.io/platform-support/#relationship:embedded) for a list of hardware that will support what you want to do.

zhou1591, 2023-10-08:
> Hello @zhou1591 , As far as I know you always need a computer to be able execute nodeJS in order to be able to run johnny-five. It is not code that you can compile and put on Arduino with a battery. If you want something compact you'll need something like a raspberry pi zero in addition to your Arduino to execute johnny-five. This question is a duplicate of this topic #1792

thanks

zhou1591, 2023-10-08:
> This requires an on-board linux environment. See the [Platform Support page](https://johnny-five.io/platform-support/#relationship:embedded) for a list of hardware that will support what you want to do.

thanks

zhou1591, 2023-10-10:
> Hello @zhou1591 , As far as I know you always need a computer to be able execute nodeJS in order to be able to run johnny-five. It is not code that you can compile and put on Arduino with a battery. [...]

I now have a nodemcu, and I want to compile the Jonny-Five file and upload it to nodemcu, can I use some vscode plugins to implement it?

zhou1591, 2023-10-10:
> This requires an on-board linux environment. See the [Platform Support page](https://johnny-five.io/platform-support/#relationship:embedded) for a list of hardware that will support what you want to do.

I now have a nodemcu, and I want to compile the Jonny-Five file and upload it to nodemcu, can I use some vscode plugins to implement it?

scottgonzalez, 2023-10-10:
There is no such thing as compiling Johnny-Five files. nodemcu is firmware that runs lua, that is not what you need. As I said, you must use a board listed on the platform support page. These boards run Node.js directly; they are the host instead of your computer being the host.