rwaldron/johnny-five #1822: Cannot Input Mode Pin 12, 14, 15. ESP8266

YovanggaAnandhika, 2023-05-22:
Device Type : Lolin V3 (ESP8266 Wifi)
````Typescript

let board = new Board({ debug : false, repl : false, port : "COM5"});

board.on("ready", async () => {

     const Up = new Relay({ pin : 4 });
     const Button1 = new Button({ pin : 2 }); // Pin D4
     const Button2 = new Button({ pin : 13 }); // pin D5

     await Ticket.on("down", async () => {
                //await Up.close();
     });

      await Button1.on("up", async () => {
              await Up.open();
              await setTimeout(async () => {
                  await Up.close();
              }, 500)
       }); // its Work

      await Button21.on("up", async () => {
              await Up.open();
              await setTimeout(async () => {
                  await Up.close();
              }, 500)
       }); // Not Work

});

````

moeinrahimi, 2023-07-27:
hi I don't know the answer to your question but I want to know doest relay works fine? did you use the firmataWifi from arduino ide example?

YovanggaAnandhika, 2023-08-01:
the relay functions properly when using the out mode. 
but, for the input mode can not respond to the input signal @moeinrahimi 

hardware : [NodeMCU Lolin v3](https://components101.com/development-boards/nodemcu-esp8266-pinout-features-and-datasheet)
C++ Ino : [FirmataPlus](https://github.com/YovanggaAnandhika/DKAFramework-Typescript/blob/production/packages/IoT/lib/StandardFirmataPlus.ino)

moeinrahimi, 2023-08-07:
so you are using firmataplus not firmatawifi?

YovanggaAnandhika, 2023-12-21:
> so you are using firmataplus not firmatawifi?

Yes, i am using control via usb communication.