rwaldron/johnny-five #1825: Usage with Electron of latest J5 versions?

Pimentoso, 2023-06-06:
Hello,

I maintain an Electron app that uses J5 to control an arduino board via usb. I need to upgrade the node version but the ancient Electron I'm using (9.x) is no good, so I need to upgrade it and it brings along J5+serialport as well.

But I keep getting the `Loading non-context-aware native module in renderer` error
```
Uncaught Error: Loading non-context-aware native module in renderer: '/xxx/electron-johnny-five-examples/1-led/node_modules/@serialport/bindings/build/Release/bindings.node'. See https://github.com/electron/electron/issues/18397.
    at process.func [as dlopen] (node:electron/js2c/asar_bundle:5:1812)
    at Object.Module._extensions..node (node:internal/modules/cjs/loader:1199:18)
    at Object.func [as .node] (node:electron/js2c/asar_bundle:5:1812)
    at Module.load (node:internal/modules/cjs/loader:988:32)
    at Module._load (node:internal/modules/cjs/loader:829:12)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13343)
    at Function.o._load (node:electron/js2c/renderer_init:33:356)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at bindings (/xxx/electron-johnny-five-examples/1-led/node_modules/bindings/bindings.js:112:48)
```

Which is raised by serialport, but a minimal app using Electron+serialport (no J5) works fine (Electron 17 and serialport 10.5.0) --> https://github.com/serialport/electron-serialport

Here is a minimal app using Electron+J5 demonstrating the error, if anyone wants to replicate
https://files.catbox.moe/tlezc1.zip
It's adapted from an example that can be found here https://github.com/sofroniewn/electron-johnny-five-examples (not maintained anymore)

The versions I'm aiming to upgrade are
- node 18
- electron 17
- J5 2.1.0
- serialport 10.5.0

Pimentoso, 2023-06-16:
I temporarily solved this by using johnny-five directly from the repo, since serialport has been updated to 10.5 which is N-API compliant
```
    "johnny-five": "rwaldron/johnny-five",
```
And now electron does not complain about native modules in renderer.

I would still like a full example of johnny-five usage in electron with proper context isolation.

ikcsup, 2023-06-19:
Hi @Pimentoso ,

could you please explain to a newby how to solve this? I'm trying to simply test led blinking so i can go on with the development, but i'm seriously stuck with this problem.

Thanks a lot.

Pimentoso, 2023-06-20:
Sure @ikcsup I will try making the led blink example work and push it on a repo so you can see the whole thing.

Pimentoso, 2023-06-22:
Sorry, still not working I'm afraid. It started raising the error again as soon as I connected the arduino to the USB.

[image]

JFigueroa51, 2023-12-21:
@Pimentoso, did you ever find a way around this? I'm looking to make a simple app with J5 and I was hoping to use electron for the UI, but I don't want to use really old versions of electron if possible

Pimentoso, 2023-12-21:
@JFigueroa51 nope sorry. In the end I kept using Electron 9 with Node 12 and updated all the libraries to the latest supported versions. It's a shame because I can't natively build a release for ARM macbooks anymore.
Also J5 doesn't look maintained anymore and I'm not good enough at node to fork and fix it myself.