rwaldron/johnny-five #1831: INPUT_PULLUP on digital input pin

Dave645, 2023-09-21:
Hello,

I see all over the place to use pinMode(pinNumber, INPUT_PULLUP);

How do I activate INPUT_PULLUP with johnny-five?

I have an digital input that requires a resistor and the board should contain some built-in pullup resistors.

Thank you.

dtex, 2023-09-22:
Setting pin mode is something you would handle with the IO, so it's not a Johnny-Five thing really, but you can access the IO through the Johnny-Five Board instance:

```js
import { board } from "johnny-five";

const board = new five.Board();

board.on("ready", function() {
  board.io.pinmode(3, board.io.MODES.PULLUP);
});
```

You say that you see that example all over the place. Can you point me to that? I've seen few people ask questions like this, and I'm curious where it's all coming from.

Dave645, 2023-09-22:
I believe a lot of programmers use INPUT_PULLUP instead of regular INPUT mode, as to avoid the noise in having something similar to a floating pin.

Some examples:
- https://forum.arduino.cc/t/digital-pins-register-confusion-input_pullup-high-and-low/179087/3
- https://youtu.be/wxjerCHCEMg?t=187
- https://youtu.be/OZGMLOwHYf8?t=251

Dave645, 2023-09-22:
To clarify. I am not using it for a button. I am using it and almost all other digital pins to read from an external source.

dtex, 2023-09-22:
Can you share your code?

Dave645, 2023-09-22:
> Can you share your code?

```js
const five = require("johnny-five");

const board = new five.Board();

board.on("ready", function () {
  board.io.pinMode(31, board.io.MODES.PULLUP);

  const sensor = new five.Sensor.Digital({
    pin: 31,
  });

  sensor.on("change", function () {
    console.log("Pin value", this.value);
  });
});
```

The thing is the wire connecting to that pin will sometimes have to be "loose" making it a floating pin. And I need the resistor to "filter out" all the noise.

dtex, 2023-09-22:
I can't test (I'm at work), but I suspect initializing a sensor on that pin will reset it to a normal input. Have you tried setting the pin mode after instantiating the sensor?

Dave645, 2023-09-22:
It doesn't seem to work even with just using pinInput.read()

ScreamZ, 2023-10-23:
In my experience, I've been able to proceed using the Button library https://github.com/rwaldron/johnny-five/blob/main/docs/button-pullup.md

`isPullup: true` This has been working for some time on my codebase. I've been able to test it 2 days ago and it was working but when i switched back to the code related to this part, it now glitch

```ts
 button.on("up", function() {
    led.off();
  });
```

triggers constantly, so this is not fixed. The doc seems to write that yes, but when you read at the code, all it it doing is sending a write command to LOW on that pin…

tbaltrushaitis, 2023-10-25:
I should mention that I had some problems that looks like similar to this one in the topic.
But finally we're figured out that root cause of the problem was in failed arduino pin.
I've rechecked this a couple of times and problem with pin is now **confirmed**

tbaltrushaitis, 2023-10-25:
I suggest in case of any troubles to check all of the modes of a button: `isPullup`, `isPulldown` and `invert` and play with every of those modes to be sure.

ScreamZ, 2023-10-25:
> I should mention that I had some problems that looks like similar to this one in the topic. But finally we're figured out that root cause of the problem was in failed arduino pin. I've rechecked this a couple of times and problem with pin is now **confirmed**
U mean your arduino is damaged ?

> I suggest in case of any troubles to check all of the modes of a button: `isPullup`, `isPulldown` and `invert` and play with every of those modes to be sure.

What is is supposed to change ? Its only server code and if its related to the pin physically I'm not sure things will

tbaltrushaitis, 2023-11-06:
> > U mean your arduino is damaged ?
Yes

> What is is supposed to change ? Its only server code and if its related to the pin physically I'm not sure things will
What you mean in "server code"?