rwaldron/johnny-five #1833: Input pull-up doesn't work on expanders

scottgonzalez, 2023-10-05:
I'm using an MCP23017, which supports pull-up resistors on all pins. When initializing a button with `isPullup: true`, the pull-up resistor is not activated on the pin.

scottgonzalez, 2024-01-09:
I found a way to get this working. Here's more context and some follow-up questions:

I expected the following code to work:

```js
    const expander = new five.Expander({
      controller: "MCP23017",
      address: 0x21,
    });
    const virtual = new five.Board.Virtual(expander);

    const button = new five.Button({
      board: virtual,
      isPullup: true,
      pin: 5,
    });
```

However, this only configures the pin to be an input pin, without the internal pull-up resistor enabled. Looking through the code, I saw that the MCP23017 code does have support for controlling the pull-up registers and I was able to get the button to work with the following code:

```js
    const expander = new five.Expander({
      controller: "MCP23017",
      address: 0x21,
    });
    const virtual = new five.Board.Virtual(expander);

    // ADDED COMMAND
    expander.pullUp(5, virtual.io.HIGH)

    const button = new five.Button({
      board: virtual,
      isPullup: true,
      pin: 5,
    });
```

So I have two questions:
1. Is the expectation that the first block of code should have worked?
2. If the button API alone is not expected to work, is my current approach correct or am I reaching too far into the internals?

tbaltrushaitis, 2024-05-31:
@scottgonzalez , 

I think if you got the workaround and its really work and makes you happy then don't ask unnecessary questions :-)

p.s. thank you for sharing your solution! 👍🏻

scottgonzalez, 2024-06-01:
These are not unnecessary questions. Having to use a workaround does not make me happy and I believe this is a bug in J5.