rwaldron/johnny-five #1841: RaspberryPi issue

technic404, 2024-02-17:
I'm using Raspberry Pi 4B.

code:
```js
var five = require("johnny-five");
var board = new five.Board();

board.on("ready", function() {
    console.log('Ready!');
});
```

Error:
```
internal/deps/acorn/acorn/dist/acorn.js:4015
      (ch >= 0x41 /* A */ && ch м쀽 0聸耵C  *РØ 肪࠯, ɼ|逊 Ȩ   0䀨䡣jĠ>= 0x¶1 /ЪĠá */(&₦耠䁣聨ࠨ¼耿 8¸‷䉁Ƞ/쀪ဠࡺ */)
                                ^^

SyntaxError: Unexpected identifier

at Object.<anonymous> (/home/user/Desktop/node_modules/johnny-five/lib/repl.js:2:14)
```

As i was testing the issue, error is displayed in the first line (`var five = require("johnny-five");`).

scottgonzalez, 2024-03-05:
I have no issues with my Pi 400s.

But the provided error code looks more like a broken source file than an issue with Johnny-Five. It should be impossible for a Johnny-Five error to produce a `SyntaxError`.

lbineau, 2024-03-29:
In addition you can try to install the main branch instead of npm (npm last update is 3years ago, while github is 1year ago).
https://github.com/rwaldron/johnny-five?tab=readme-ov-file#source-code

PizzaProgram, 2025-02-01:
> (npm last update is 3years ago, while github is 1year ago)

But the Release version number here matches the one at `npm` : 2.1.0  
I think @johnny-five would update on npm too, if anything would change.

_IMHO this issue can be closed._

PizzaProgram, 2025-02-01:
@scottgonzalez I take it back. I see now that [Serial has been updated](https://github.com/rwaldron/johnny-five/pull/1808) in 2023, but no new release has been presented.
So you are right, the source downloaded directly from here (and not from "release" page) is newer than the one on npm.

Maybe we should ask to make a new `2.2.20250201` release ? 
_(it is always a good practice to include a date in the numbering)_

scottgonzalez, 2025-02-01:
I wouldn't hold your breath for a release. I gave up waiting for one a long time ago and started using my own fork (also forked `firmata` to use an updated `serialport` which gets pulled into `johnny-five`). I point to the tip of https://github.com/scottgonzalez/johnny-five/tree/blue-owl (currently using `"johnny-five": "scottgonzalez/johnny-five#f548f9aed5b1992cb8f847e95b5a4583f952fc73"` as my dependency) for my projects.

As for using a date as a patch version, that is definitely not a good practice. If you want to include a date in the version, it should be in the build metadata, e.g., `2.2.0+20250201`, but I personally would not recommend that either.

@dtex can you close this issue?

dtex, 2025-02-01:
Actually, I should ping @rwaldron again. I think he believes I can publish releases at npm, but that's the one privilege I don't have.

PizzaProgram, 2025-02-02:
@scottgonzalez  Actually my real goal is to find a stable Node-Red module, that would properly handle Firmata.  
 _(Currently I'm using [RPi-Pico](https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html) boards with [Configurable-Firmata](https://github.com/firmata/ConfigurableFirmata) firmware on them.)_ 

The Node-Red's built-in module called: **35-aruino**, based on [Firmata-io](https://github.com/yokobond/firmata-io),  can not properly handle: 
 - Re-Deploy / 
 - USB-disconnect /
 - reconnect / 
 - wrong pin number /  << the whole NR is crashing
 - (temp reading)  

The whole Arduino module is freezing down. I've checked the codebase of those files, and I think the main problem is:
- there is not any `serial.on('error')...` present, 
- no timeout of data flow
- nor enough try..catch .

So I thought what if Johnny-five can handle it better? _(Was big fan of the movie as a child 🤖  .)_

tbaltrushaitis, 2025-03-07:
> Was big fan of the movie as a child 🤖

💯