rwaldron/johnny-five #1844: LILYGO SIM7000G - Issue using PWM controlling a DC motor and a servo at the same time

trond11907, 2024-10-02:
Hi there..

I have had some code working for a long time where I used a LILYGO SIM7000G unit connected to a DRV8871 motor driver through pins 32 and 33, and a servo connected to PIN 12. But suddenly/lately it has started to fail and the servo is pushed all the to one end as soon as I do a analogWrite to PIN 32 or 33. I have not done any changes that should have affected this. I also was able to reproduce the problem with a small test program.

#include <ESP32Servo.h>
Servo myservo;

#define SerialAT Serial1

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200); 
  myservo.attach(12, 1000, 2000); 
  pinMode(32, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("Loop...");
  myservo.write(0);
  delay(2000);     
  myservo.write(180);
  delay(2000);

//If I comment out the line below, the servo works fine "every loop", 
//but the servo only works the first round when this line is active.
  analogWrite(32, 50);  //This starts the dc motor but also pushes the servo to the end and it stops working.
  delay(2000);
}

Any ideas folks? :)

dtex, 2024-10-02:
This is not Johnny-Five code. If it was working fine and then just started acting weird after no changes, I'd be inclined to look for a wiring issue or a short. Physically disconnect the motor controller connection. Does the servo still go wonky when you send a signal to pin 32?

trond11907, 2024-10-08:
Hi... sorry for the late respons!  The strange thing is that I get the same issue whether the motor driver is connected or not to pins 32, and also if I use pin 13 instead of 12 for the servo. As soon as I call the analogWrite(32,50); line, the servo goes all the way to one end and stops responding to the myservo.write(0) and (180) commands. If the analogWrite(32,50); line is removed, the servo continuously goes back and forth.

dtex, 2024-10-08:
Okay, new theory. The analogWrite on pin 32 is changing the clock (the pins are sharing a clock). Servos are very particular about frequency, but motor controllers are not. Try initializing the motor with analogWrite(32, 0) and then initialize the servo.

dtex, 2024-10-08:
Cool... but note there is a catch. Your motor will be operating on a PWM frequency of 50hz. If this were an LED, you might be able to perceive a flicker. I don't ~think~ it matters, but it wouldn't hurt to check the motor controller documentation and make sure they don't specify an acceptable range for the PWM frequency. For most simple H-Bridges I really doubt it's an issue, but I wouldn't want you to cook anything.